MoonModules/WLED-MM #209: PIR Sensor doesn't work when using a 2D Matrix with the web interface closed

AchillesPDX, 2025-01-02:
### What happened?

The PIR sensor stops working completely when using a 2D matrix if you close the web interface or disable the GFX preview.

### To Reproduce Bug

Load WLED MM esp32_4MB_XL

Connect PIR Sensor to pin 32

Set up a 2D Matrix of pixels

Attempt to use a PIR Sensor with the web interface open and the GFX preview enabled - should work fine.

Attempt the same thing but this time with the GFX preview disabled or the web interface closed - doesn't work.

### Expected Behavior

I would expect the PIR Sensor to continue working, just as it does if it's not a matrix.

### Install Method

From https://wled-install.github.io/

### What version/release of MM WLED?

WLEDMM 14.5.1-dev (build 2412300)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2025-01-03:
Should be solved by fdc43e0a2b966e3ecce05f3868022b774024f1c9 - please try again with build 2501030 (or newer).

AchillesPDX, 2025-01-04:
> Should be solved by [fdc43e0](https://github.com/MoonModules/WLED/commit/fdc43e0a2b966e3ecce05f3868022b774024f1c9) - please try again with build 2501030 (or newer).

Thank you so much! Seems to be working now. *whew*