MoonModules/WLED-MM #212: Microphone is not accepted

NickDolfin, 2025-01-07:
### What happened?

I am using visual studio code and platformio: env:32S3_8MB_M (WLED-14.5.0-beta.

My microphone is a ICS-43434. Similar to the obsolete INMP441.
Using the older moonmodule software firmwares there is no problem.

The main issue is right now that it is using UDP sound sync when I look at the info panel.
Why??

### To Reproduce Bug

[image]
[image]

### Expected Behavior

I have looked at all the settings but I can't find any setting that swiches it to a microphone.

Anyone any idea?
Thanks a lot!

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

v14.5.0-beta "Small Step" WLEDMM esp32S3_8MB.bin build 2412180

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2025-01-07:
@NickDolfin it might be that audioreactive is in sound receive mode - it means that the microphone is overruled by network input.

Please scroll down on the audioreactive settings page, and check that "Sync Mode" is set to "Off".

If you still have the problem, please post your custom build env, plus your cfg.json so I can try to reproduce the problem.

softhack007, 2025-01-07:
ps: looking at your settings - gain = 250 and AGC=off is quite brutal for the ICS-43434. 
This mic works best with AGC=lazy, gain = 30, squelch = 5 (maybe squelch=10 if you still receive noise at 5).

NickDolfin, 2025-01-07:
Wow Softhack,

You are absolutely right. 
I missed that setting.

And thanks for the settings info. Works quite nice.

Now, I can use this fantastic software and play around with it. 

Many thanks !!

I have also some questions about the filing system that does not accept zip or pdf files, but I will save that for another thread.
And also something about the source - in which files can I change the (for me) wrong default io pins and other default settings?

closed 2025-01-07 by NickDolfin