MoonModules/WLED-MM #222: Reboot Adafruit Matrix Portal S3

letmoc, 2025-02-01:
### What happened?

Occasional reboot of the Adafruit Matrix Portal S3 board.

### To Reproduce Bug

The Adafruit Matrix Portal S3 works with or without a display for about 24-48 hours. Then the board restarts.

### Expected Behavior

no reboot

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

14.5.0-beta WLEDMM_14.5.0-beta_adafruit_matrixportal_esp32s3.bin & build 2501170 firmware-adafruit_matrixportal_esp32s3_tinyUF2.bin

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell

```

### Anything else?

This issue has been moved from the discussion 'Problem with installing WLEDMM on Adafruit Matrix Portal S3.' @softhack007 participated in the discussion.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

letmoc, 2025-02-01:
Assigned a static IP address, enabled/disabled NTP. Operated the WF2 board in parallel for comparison; the WF2 board runs continuously.

netmindz, 2025-09-20:
Can't help without logs showing the reboot