MoonModules/WLED-MM #243: Inability to set a boot preset

cyberdog-17, 2025-06-20:
### What happened?

When changing the default boot preset from 0 to 1, or any other preset configured, the Apollo Matrix-1 will crash and boot to an orange colored matrix.

### To Reproduce Bug

1. Navigate to LED preferences
2. Go to "defaults"
3. Where it says "Apply preset [0] at boot (0 uses default)" change that number to 1 (or any other preset number that the device has saved).
4. Notice the device crash and reboot to the orange colored matrix.
5. Navigate back to LED preferences
6. Go back to defaults
7. Notice the number previously set is now back at 0 

### Expected Behavior

When changing the default boot preset from 0, I expect the device to successfully save the preset number and apply that preset at device boot instead of the system default of 0. It is also expected that the device not crash when changing this default. 

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

v0.14.1 -b32.40 "Fried Chicken"

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420a1932  PS      : 0x00060330  A0      : 0x82069b12  A1      : 0x3fcebd20
A2      : 0x00000054  A3      : 0x00000000  A4      : 0x80000020  A5      : 0x00000007
A6      : 0x00000005  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000020
A10     : 0x3fca07d0  A11     : 0x00000000  A12     : 0x00001000  A13     : 0x00000040
A14     : 0x3fca0730  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000090  LBEG    : 0x40382b24  LEND    : 0x40382b32  LCOUNT  : 0x00000001

Backtrace: 0x420a192f:0x3fcebd20 0x42069b0f:0x3fcebd50 0x4200b74e:0x3fcebd70 0x4200b7c9:0x3fcebd90 0x4202278d:0x3fcebdb0 0x420227a4:0x3fcebdd0 0x420227b1:0x3fcebdf0 0x42022819:0x3fcebe10 0x420558bc:0x3fcebe30 0x42055a2f:0x3fcebe60 0x4206fc59:0x3fcebe80

ELF file SHA256: 21b13a12378f7602

E (16324) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420d7803
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x28dc
entry 0x403c98c0
E (324) esp_core_dump_flash: No core dump partition found!
E (324) esp_core_dump_flash: No core dump partition found!
E (327) psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (664) I2S: i2s_driver_uninstall(2048): I2S port 0 has not installed
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420a1932  PS      : 0x00060b30  A0      : 0x82069b12  A1      : 0x3fcebd20
A2      : 0x00000054  A3      : 0x80000001  A4      : 0x803828f1  A5      : 0x3fccdec0
A6      : 0x00060023  A7      : 0x00000003  A8      : 0x00060023  A9      : 0x80000000
A10     : 0x02c99610  A11     : 0x00ffffff  A12     : 0x80379da7  A13     : 0x3fccde90
A14     : 0x00000001  A15     : 0x3fcea000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000090  LBEG    : 0x40382b24  LEND    : 0x40382b32  LCOUNT  : 0x00000001

Backtrace: 0x420a192f:0x3fcebd20 0x42069b0f:0x3fcebd50 0x4200b74e:0x3fcebd70 0x4200b7c9:0x3fcebd90 0x4202278d:0x3fcebdb0 0x420227a4:0x3fcebdd0 0x420227b1:0x3fcebdf0 0x42022819:0x3fcebe10 0x420558bc:0x3fcebe30 0x42055a2f:0x3fcebe60 0x4206fc59:0x3fcebe80

ELF file SHA256: 21b13a12378f7602

E (5942) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4210142e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x28dc
entry 0x403c98c0
E (324) esp_core_dump_flash: No core dump partition found!
E (324) esp_core_dump_flash: No core dump partition found!
E (327) psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (664) I2S: i2s_driver_uninstall(2048): I2S port 0 has not installed
```

### Anything else?

The Controller I am using is the Apollo M-1 Matrix display with the hub75e panel

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

softhack007, 2025-10-20:
@cyberdog-17 the console logs contain a line "PSRAM chip not found or not supported".
Can you double-check that the firmware you uploaded is the correct one for your PSRAM configuration? Also please note that the bootloader and WLED-MM firmware must match exactly; if you used OTA to upgrade from vanilla WLED to WLED-MM this is very often leading to problems later.
@lost-hope might be able to help you in case you're not sure what firmware is right for the board.

About the reboot issue: I have never experienced this in my tests (normally I use a boot preset on my boards). I'll try if I can reproduce it here.

netmindz, 2025-11-09:
https://github.com/netmindz/WLED-MM/blob/Apollo_M-1/platformio_override.ini